IntelligentRoboticsLabs/gb_visual_detection_3d #10: how do I get  map-based X, Y, Z coordinates of detected object by darknet ros.

hyunoklee, 2019-09-16:
Hello. I'm very interested the darknet ros 3d package.
I have a question about the package.
I place depth camera on turtlebot3. And tutlebot3 navigate room .  
And I want to get map-based X, Y, Z coordinates of darknet ros detected object using the your packages.
Do I just modify the value of working_frame: in the darknet_3d.yaml file as below?
working_frame:/camera_link  -> working_frame:/map . 

If you answer me, very thanks you.

fmrico, 2019-09-17:
Hi @hyunoklee 

This software has one limitation: you must use a framework whose X-axis is oriented towards the scene.

I would use base_footprint as the working frame, and transform the center to "map", with something like this (I have not compiled this!!!, use it only as reference):

```
// The input is BoundingBoxes3d msg;

BoundingBox3d point_msg = msg[0];
tf2::Vector3 point;

point.setX( (point_msg.xmax + point_msg.xmin) / 2.0 );
point.setY( (point_msg.ymax + point_msg.ymin) / 2.0 );
point.setZ( (point_msg.zmax + point_msg.zmin) / 2.0 );

geometry_msgs::TransformStamped p2map_msg;
tf2::Transform p2map;

std::string error;
if (tfBuffer_.canTransform(point_msg.header.frame_id, "map", ros::Time(0), ros::Duration(0.1), &error))
	p2map_msg = tfBuffer_.lookupTransform(point_msg.header.frame_id, "map", ros::Time(0));
else
{
        ROS_ERROR("Can't transform %s", error.c_str());
	return;
}

tf2::convert(p2map_msg, p2map);

tf2::Vector3 point_in_map = (p2map).inverse() * point;

```
Hope it helps!!!

closed 2019-09-27 by fmrico

gokcesena, 2023-11-27:
Hi @fmrico, in which part should I integrate this code? Darknet3d.cpp or darknet3DListener.cpp?

fmrico, 2023-11-27:
@fgonzalezr1998 ?

fgonzalezr1998, 2023-11-28:
Hello, @gokcesena . I supose you have a node which subscribes to the darknet_ros_3d output. You only need to add some code to that node in order to transform the x,y,z coordinates to the map frame. You should use tf or tf2.

To sum up, the coordinates of the 3d bounding boxes can be changed to any frame AFTER their calculation (many documentation about coordinates transformation is available on internet), so don't change de working_frame parameter of the yaml file.