IntelligentRoboticsLabs/gb_visual_detection_3d #24: Waiting for image.

santarone, 2020-04-08:
Hello.
Thank you for sharing the great project.

There are a few problems.

First, I installed all the necessary elements in advance.
Darknet_ros also works normally.

I use a model that I learned personally.
This includes hammer pliers, etc.

my specification

Ubuntu 18.04 LTS
YOLO v3
Intel D435
Ross Melodic
That's it.

First

`roscore`

Run

`rolaunch realense2_camera rs_camera.launch filters:=pointcloud`

on it.

`rorun rviz rviz`

It's been launched.

After this

`rolaunch darknet_ros_3d darknet_ros_3d.launch`

when

Waiting for image. It appears consecutively.
What should I do?
Please give me some advice.

Attached are the parameters and the rostopic list.

PARAMETERS
* /actions/camera_reading/name: /darknet_ros/check...
* /darknet_3d/darknet_ros_topic: /darknet_ros/bounce...
* /darknet_3d/interpreted_classes: ['player', 'hammer...
* /darknet_3d/minimum_probability: 0.3
* /darknet_3d/minimum_detection_themeshold: 0.3
* /darknet_3d/output_bbbx3d_topic: /darknet_ros_3d/b...
* /darknet_3d/point_cloud_topic: /camera/dept_reg...
* /darknet_3d/working_frame: camera_link
* /darknet_ros/config_path: /home/gilen/catk...
* /darknet_ros/subscribers/camera_reading/topic: /camera/rgb/image...
* /darknet_ros/weights_path: /home/gilen/catk...
* /darknet_ros/yolo_model/config_file/name: p.cfg
* /darknet_ros/yolo_model/detection_classes/names: ['hammer', 'plier...
* /darknet_ros/yolo_model/threshold/value: 0.3
* /darknet_ros/yolo_model/weight_file/name: p_final.weights
* /image_view/enable_console_output: True
* /image_view/enable_opencv: True
* /image_view/wait_key_delay: 1
* /publishers/bounding_boxes/latch: False
* /publisher/bounding_boxes/queue_size: 1
* /publishers/bounding_boxes/topic: /darknet_ros/bounce...
* /publics/detection_image/latch: True
* /publics/detection_image/queue_size: 1
* /publics/detection_image/topic: /darknet_ros/detet...
* /publics/object_detector/latch: False
* /publics/object_detector/queue_size: 1
* /publishers/object_detector/topic: /darknet_ros/foun...
* /rosdistro: melodic
* /rosversion: 1.14.3

`rostopic list`
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth_registered/points
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/image_rect_raw/theora
/camera/infra2/image_rect_raw/theora/parameter_descriptions
/camera/infra2/image_rect_raw/theora/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb/image_raw
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/clicked_point
/darknet_ros/bounding_boxes
/darknet_ros/check_for_objects/cancel
/darknet_ros/check_for_objects/feedback
/darknet_ros/check_for_objects/goal
/darknet_ros/check_for_objects/result
/darknet_ros/check_for_objects/status
/darknet_ros/detection_image
/darknet_ros/found_object
/darknet_ros_3d/bounding_boxes
/darknet_ros_3d/markers
/diagnostics
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

raghukamath, 2020-04-08:
Hi @santarone ,

I am working with same setup as you i.e:
Ubuntu 18.04 LTS
YOLO v3
Intel D435
Ross Melodic

I do not have the "waiting for image" error as I modified the Config camera image topic in darknet_ros_3d.lauch to this:
  <arg name="camera_rgb_topic" default="/camera/color/image_raw" />

I can visualize the bounding boxes but I can't see the 3D point cloud in RVIZ.

santarone, 2020-04-09:
Hi sir! 
i solved my problem. subscriber was problem.

in Darknet_ros_3d.launch

```
<!-- Config camera image topic  -->
  <arg name="camera_rgb_topic" default="/camera/color/image_raw" />
```

any way. if you want display cloud point 2 at rviz via intelrealsense 
try launch 

`$ roslaunch realsense2_camera rs_rgbd.launch filters:=pointcloud`

before darknet_3d launch

and 
at rviz Add -> By topic 
/depth -> /color -> /points -> PointCloud2

I hope it works to you as i do.
thanks for comment.

have a great day!

fmrico, 2020-04-09:
Great @santarone !!

And thanks @raghukamath ;)

closed 2020-04-09 by fmrico