IntelligentRoboticsLabs/gb_visual_detection_3d #25: How to track the 3D bounding boxes?

mhaboali, 2020-05-28:
Hi @fgonzalezr1998 

I managed to run the system successfully with RealSense D435i and it works perfectly way better than expected. Now I'm looking for adding 3D bounding boxes tracking not only detection, to achieve more robustness. Is that doable?

I'm willing to contribute to implementing this feature for your pkg if it doesn't have yet. 

Thanks!

fgonzalezr1998, 2020-05-28:
You can visualize the 3d bounding boxes adding MarkerArray in *rviz*. 
Objects tracking is something we have implementing and we hope to be ready in September, more or less. The most interesting thing is to predict possible paths that mobile objects can do, not only to trace these paths. This is exactly what I am working on and, I hope this tool to be ready in September, approximately.
There is no doubt that this tool would be great. Thank you for your contribution!

mhaboali, 2020-05-29:
Awesome to hear that!

Thanks so much for this information and good luck 🤞

closed 2020-05-29 by mhaboali