IntelligentRoboticsLabs/gb_visual_detection_3d #34: Error: no matching function for call to 'tf2_ros::Buffer::lookupTransform'

jameslele, 2020-09-17:
I got a problem when i run **colcon build**
System: Ubuntu 18
~/dev_ws$ colcon build --symlink-install
Starting >>> darknet_ros_msgs
Starting >>> gb_visual_detection_3d_msgs
Finished <<< gb_visual_detection_3d_msgs [0.40s]                                                                                          
Finished <<< darknet_ros_msgs [0.45s]
Starting >>> darknet_ros
Starting >>> darknet_ros_3d
                                                                                 
--- stderr: darknet_ros_3d                                                                        
/home/jiale/dev_ws/src/gb_visual_detection_3d/darknet_ros_3d/src/darknet_ros_3d/Darknet3D.cpp: In member function ‘void darknet_ros_3d::Darknet3D::update()’:
/home/jiale/dev_ws/src/gb_visual_detection_3d/darknet_ros_3d/src/darknet_ros_3d/**Darknet3D.cpp:216:61: error: no matching function for call to** ‘tf2_ros::Buffer::lookupTransform(std::__cxx11::string&, std_msgs::msg::Header_<std::allocator<void> >::_frame_id_type&, std_msgs::msg::Header_<std::allocator<void> >::_stamp_type&, tf2::Duration)’
         point_cloud_.header.stamp, tf2::durationFromSec(2.0));

fgonzalezr1998, 2020-09-18:
Hi, @jameslele Have you installed tf2_ros? You can type ``sudo apt-get install ros-eloquent-tf2-ros`` or you can clone [this repo](https://github.com/ros2/geometry2/tree/eloquent). It is working for me and I have looking the code of the repo and *lookupTransform* exists, so probably you have any problem with tf2_ros library. Try the two options I said and ask me what you need. Your feedback is very important. Thank you.

closed 2020-10-02 by jameslele

miga94, 2022-11-30:
Hello, I am obtaining following error i have ros-dashing ubuntu18.04 in Roscube-X. I have installed ros-dashing-tf2-ros.
/home/robolab/ros2_ws/gb_visual_detection_3d/darknet_ros_3d/src/darknet_ros_3d/Darknet3D.cpp:216:61: error: no matching function for call to ‘tf2_ros::Buffer::lookupTransform(std::__cxx11::string&, std_msgs::msg::Header_<std::allocator<void> >::_frame_id_type&, std_msgs::msg::Header_<std::allocator<void> >::_stamp_type&, tf2::Duration)’
         point_cloud_.header.stamp, tf2::durationFromSec(2.0));
                                                             ^
In file included from /home/robolab/ros2_ws/gb_visual_detection_3d/darknet_ros_3d/include/darknet_ros_3d/Darknet3D.hpp:22:0,
                 from /home/robolab/ros2_ws/gb_visual_detection_3d/darknet_ros_3d/src/darknet_ros_3d/Darknet3D.cpp:18:
/opt/ros/dashing/include/tf2_ros/buffer.h:79:5: note: candidate: virtual geometry_msgs::msg::TransformStamped tf2_ros::Buffer::lookupTransform(const string&, const string&, const TimePoint&, tf2::Duration) const
     lookupTransform(const std::string& target_frame, const std::string& source_frame,
     ^~~~~~~~~~~~~~~
/opt/ros/dashing/include/tf2_ros/buffer.h:79:5: note:   no known conversion for argument 3 from ‘std_msgs::msg::Header_<std::allocator<void> >::_stamp_type {aka builtin_interfaces::msg::Time_<std::allocator<void> >}’ to ‘const TimePoint& {aka const std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >&}’
/opt/ros/dashing/include/tf2_ros/buffer.h:96:5: note: candidate: virtual geometry_msgs::msg::TransformStamped tf2_ros::Buffer::lookupTransform(const string&, const TimePoint&, const string&, const TimePoint&, const string&, tf2::Duration) const
     lookupTransform(const std::string& target_frame, const tf2::TimePoint& target_time,
     ^~~~~~~~~~~~~~~
/opt/ros/dashing/include/tf2_ros/buffer.h:96:5: note:   candidate expects 6 arguments, 4 provided
In file included from /opt/ros/dashing/include/tf2_ros/buffer_interface.h:37:0,
                 from /opt/ros/dashing/include/tf2_ros/buffer.h:35,
                 from /home/robolab/ros2_ws/gb_visual_detection_3d/darknet_ros_3d/include/darknet_ros_3d/Darknet3D.hpp:22,
                 from /home/robolab/ros2_ws/gb_visual_detection_3d/darknet_ros_3d/src/darknet_ros_3d/Darknet3D.cpp:18:
/opt/ros/dashing/include/tf2/buffer_core.h:152:5: note: candidate: geometry_msgs::msg::TransformStamped tf2::BufferCore::lookupTransform(const string&, const TimePoint&, const string&, const TimePoint&, const string&) const
     lookupTransform(const std::string& target_frame, const TimePoint& target_time,
     ^~~~~~~~~~~~~~~
/opt/ros/dashing/include/tf2/buffer_core.h:152:5: note:   candidate expects 5 arguments, 4 provided
/opt/ros/dashing/include/tf2/buffer_core.h:135:5: note: candidate: geometry_msgs::msg::TransformStamped tf2::BufferCore::lookupTransform(const string&, const string&, const TimePoint&) const
     lookupTransform(const std::string& target_frame, const std::string& source_frame,
     ^~~~~~~~~~~~~~~
/opt/ros/dashing/include/tf2/buffer_core.h:135:5: note:   candidate expects 3 arguments, 4 provided
make[2]: *** [CMakeFiles/darknet_ros_3d.dir/src/darknet_ros_3d/Darknet3D.cpp.o] Error 1
make[1]: *** [CMakeFiles/darknet_ros_3d.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< darknet_ros_3d [22.0s, exited with code 2]

heshanchuan, 2023-07-11:
I have the same question.How do you solve it?