IntelligentRoboticsLabs/gb_visual_detection_3d #35: Can we use this package with ROS (kinetic)

hsaleem1, 2020-10-04:
Hi there,

Can anybody please guide me that can I use this package with ROS (kinetic) or is it necessary to install and run this package with ROS 2?

Thanks for the help.

fgonzalezr1998, 2020-10-06:
Hi @hsaleem1 , if you want to use this package under ROS Kinetic, you only have to clone the 'melodic' branch. It will work fine. Remember cloning [the messagess](https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d_msgs) too

hsaleem1, 2020-10-09:
@fgonzalezr1998, thanks for your response.

Can you please tell me that how can I clone the melodic branch?

Thanks.

fgonzalezr1998, 2020-10-09:
@hsaleem1 Do you have other packages in your workspace? What I would do is cleaning up the workspace (removing build and devel folders) and then, type ``catkin_make``.

Make sure that you are cloning the repo into 'src' folder with ``git clone -b melodic https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d``

hsaleem1, 2020-10-12:
@fgonzalezr1998, I cloned the melodic branch and launched the node...the objects are now detecting successfully. But the 3D bounding box are not appearing. 

I edited yaml file:
darknet_ros_topic: /darknet_ros/bounding_boxes
output_bbx3d_topic: /darknet_ros_3d/bounding_boxes
point_cloud_topic: /camera/depth_registered/points
working_frame: camera_link
mininum_detection_thereshold: 0.3
minimum_probability: 0.3
interested_classes: ["person", "chair"]

I started realsense by: roslaunch realsense2_camera rs_rgbd.launch

then launched the package: roslaunch darknet_ros_3d darknet_ros_3d.launch

then rostopic list gave the following topics: 
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_color/image_raw/theora
/camera/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/theora/parameter_updates
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/aligned_depth_to_infra1/image_raw/compressed
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/compressedDepth
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/theora
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/image_rect_color
/camera/color/image_rect_color/compressed
/camera/color/image_rect_color/compressed/parameter_descriptions
/camera/color/image_rect_color/compressed/parameter_updates
/camera/color/image_rect_color/compressedDepth
/camera/color/image_rect_color/compressedDepth/parameter_descriptions
/camera/color/image_rect_color/compressedDepth/parameter_updates
/camera/color/image_rect_color/theora
/camera/color/image_rect_color/theora/parameter_descriptions
/camera/color/image_rect_color/theora/parameter_updates
/camera/color_rectify_color/parameter_descriptions
/camera/color_rectify_color/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
[... 65 lines cut ...]
  stamp: 
    secs: 1602483877
    nsecs: 612223855
  frame_id: "/camera_link"
bounding_boxes: []
---
header: 
  seq: 2
  stamp: 
    secs: 1602483877
    nsecs: 712137415
  frame_id: "/camera_link"
bounding_boxes: []
---
header: 
  seq: 3
  stamp: 
    secs: 1602483877
    nsecs: 812233980
  frame_id: "/camera_link"
bounding_boxes: []

Please guide me.

Thanks

hsaleem1, 2020-10-12:
@fgonzalezr1998, Your help is very much appreciated. Now, I can detect objects with their 3d coordinates.

Can you please guide me that how can I detect custom/other objects with this package?

fgonzalezr1998, 2020-10-12:
Hi @hsaleem1 , You can detect the object that you wants. You only have to add them to the 'interested_classes' in the configuration file.
However, this package get the 2D bounding boxes provided by darknet_ros (the neural network). It means that you will not be able to detect object that Darknet doesn't detect. If you inspect the darknet_ros package, you will see that there is a '.yaml' file that contains what classes it can detect. If you want to detect custom object that Darknet doesn't detect, you need to re-train the neural network. [Here](https://github.com/AlexeyAB/darknet) you have very detailed instructions about how to do it.

Hope that I have been helpful

hsaleem1, 2020-10-13:
@fgonzalezr1998, your guidance is much appreciated.

I will see the link which you provided.

I am closing this issue.

Thank you once again.

closed 2020-10-13 by hsaleem1