IntelligentRoboticsLabs/gb_visual_detection_3d #36: last_detection_ts_ initialization cause exception when using sim_time

AravindaDP, 2020-11-03:
https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/blob/melodic/darknet_ros_3d/src/darknet_ros_3d/Darknet3D.cpp#L64 causes out of time range exceptions when using sim_time (Supposedly due to underflow)

using ros::Time(0) to initialize this parameter should be better approach

fmrico, 2020-12-02:
Hi @AravindaDP 

Yes, you are right. This is not the right way to start this attribute. I will be happy to accept your PR :)

Maybe you should add something more for avoiding pass line 168 if nothing has been received. The idea was using the difference of time, but the situation before the first image is not so simple to resolve.

Thanks!!

Engnation, 2021-02-19:
Hello,

I am getting the errors: "**[darknet_3d-2] process has died**", and also: **"WARNING: no messages received and simulated time is active. Is /clock being published?"**  when doing a `rostopic echo` to `/darknet_ros_3d/bounding_boxes` while trying to run darknet_ros_3d within a Gazebo simulation. 

Is this related to the same issue that is being discussed above? If so, @AravindaDP have you or @fmrico been able to find a solution?

Also, if it hasn't been solved, I'd like to help, but I am not too familiar with how Darknet3D.cpp works, so I might need some additional advice / pointers  to get started.

Thanks!

Engnation, 2021-03-17:
Update: Now that I understand this problem better, I realize that I had misunderstood the original recommendation and had accidentally left the - ros::Duration(60.0) in place. By initializing `last_detection_ts_ = ros::Time(0);` the problem is now solved. Thank you @AravindaDP for this suggestion, and thank you @fmrico for providing this great ROS package!