IntelligentRoboticsLabs/gb_visual_detection_3d #40: 3d coordinates to control robotic arm

hsaleem1, 2020-11-26:
Hi there, 
Is it possible to transfer the 3d coordinates from this package to control the robotic arm for pick and place purpose?

Anybody please guide me @fgonzalezr1998.

Thanks for the help.

fgonzalezr1998, 2020-11-26:
@hsaleem1  Of course! Having the 3d bounding box, you can know the coordinate of the centre and it scale.

I have not worked with robotics arms but one thing its clear: If you hace a robotic arm, it will be publish its own TF's. So, when you have calculated the needed coordinates, you must convert it to the frame that corresponds before to send it to the arm for move it.

hsaleem1, 2020-11-27:
@fgonzalezr1998, thanks for the reply.
I will follow your instructions.

hsaleem1, 2021-02-10:
Hi @fgonzalezr1998 ,

I am getting the 3d coordinates by publishing the messages through rostopic echo.
Can you please guide me that how I can subscribe these coordinates into another package?

hsaleem1, 2021-02-14:
@fgonzalezr1998 , any response regarding the previous question please?

fgonzalezr1998, 2021-08-29:
Yes. Sorry for the late response. You have to do a ROS package with a subscriber node (using c++ or python). In the ROS Wiki you will find examples @hsaleem1 
You can reply here with your doubts about the node creation

khs-sm, 2021-10-18:
Hi @fgonzalezr1998, I also have the similar concern regarding to the coordinates of object. I'm not sure how the solution works from the previous post. Can I use lookup transform by using the frame from bounding box topic? If not, any suggestion on that so I can convert the coordinates information to the TF?