IntelligentRoboticsLabs/gb_visual_detection_3d #41: Specify number of detections

hsaleem1, 2020-11-26:
Hi there, 

Can somebody please guide me that how can we limit the number of detection after running the launch file? As the output is providing repeated and continuous values.

Thanks.

fgonzalezr1998, 2020-11-26:
Hi @hsaleem1 
This package doesn't provide a parameter for doing this. So, I can think of different things:

* Change the source code adapting it to your needs (maybe it's not the most advisable).
* To write other node which subscribes to the 3d bounding boxes and finalize when you want. Obviusly, darknet_ros_3d will follow working even if you don't need it until you kill it.
* The last option is the same that the previous one but in your custom node you can execute a "kill -9" signal over de pid of darknet_ros_3d process. In this way, your computer will not waste resources when you are not using darknet_ros_3d.

hsaleem1, 2020-11-27:
@fgonzalezr1998, can you please further guide about the third point? I mean how can I execute "kill - 9" signal?

fgonzalezr1998, 2020-12-01:
@hsaleem1 You can use the "system" function from stdlib to execute a "pkill [process_name]" (where process_name" is the process name of darknet_ros_3d). I think it will be "darknet_ros_3d_node" but you can see it with htop or ps. I recommend you to consult its man page.

Also, you can execute a fork() + exec() instead of system(). It provides a greater control over the process son but in this case may be you don't have to.