IntelligentRoboticsLabs/gb_visual_detection_3d #42: Two intel realsense cameras

hsaleem1, 2020-12-11:
Can I use two Intel Realsense cameras with this package?

fgonzalezr1998, 2020-12-11:
Why would you want to use two cameras at the same time? @hsaleem1

hsaleem1, 2020-12-12:
@fgonzalezr1998, actually I need to get 3d coordinates at different scenes.

fgonzalezr1998, 2020-12-12:
@hsaleem1  What you have to do, in this case, is making a launch file that runs two times the darknet_ros_3d_node under different namespaces. And you can pass them the parameters that you want (for example, the pointcloud topic of each camera).

[here](https://answers.ros.org/question/229185/roslaunch-namespace-substitution-within-group-tags/) you can see how to group differents nodes or launchers in namespaces in a launch file.