IntelligentRoboticsLabs/gb_visual_detection_3d #43: Does it provides support to ros1?

ranjitkathiriya, 2020-12-16:


fgonzalezr1998, 2020-12-21:
Hi @ranjitkathiriya If you use the *melodic* branch, it should work well for Noetic distro. Try it and let us know your feedback! ;)

ranjitkathiriya, 2020-12-21:
Hi @fgonzalezr1998 ,

Thanks for your reply!

I have tried the melodic branch and I am getting this error. can you help me with this, please?
ERROR:

(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/catkin_ws$ catkin_make
Base path: /home/kuka-ai/catkin_ws
Source space: /home/kuka-ai/catkin_ws/src
Build space: /home/kuka-ai/catkin_ws/build
Devel space: /home/kuka-ai/catkin_ws/devel
Install space: /home/kuka-ai/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/kuka-ai/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/kuka-ai/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /home/kuka-ai/anaconda3/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home/kuka-ai/anaconda3/bin/python3
-- Using Debian Python package layout
-- Using empy: /home/kuka-ai/anaconda3/lib/python3.8/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kuka-ai/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /home/kuka-ai/anaconda3/bin/python3 (found version "3.8.5") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 33 packages in topological order:
-- ~~  - kuka_experimental (metapackage)
-- ~~  - kuka_resources
-- ~~  - kuka_rsi_simulator
-- ~~  - moveit_kuka16_v2
-- ~~  - realsense2_description
-- ~~  - gb_visual_detection_3d_msgs (plain cmake)
-- ~~  - sara_msgs
-- ~~  - actions_tutorial
-- ~~  - pick_and_place
-- ~~  - kuka_kr10_support
-- ~~  - kuka_kr120_support
-- ~~  - kuka_kr150_support
-- ~~  - kuka_kr16_support
-- ~~  - kuka_kr210_support
-- ~~  - kuka_kr3_support
-- ~~  - kuka_kr5_support
-- ~~  - kuka_kr6_support
-- ~~  - kuka_lbr_iiwa_support
-- ~~  - darknet_ros_msgs
-- ~~  - ai_milking_ros
-- ~~  - cv_basics
-- ~~  - rvip_roi_parser
-- ~~  - realsense2_camera
-- ~~  - rvip
-- ~~  - moveit_calibration_plugins
-- ~~  - darknet_ros_3d (plain cmake)
-- ~~  - learning_tf2
-- ~~  - kuka_eki_hw_interface
-- ~~  - kuka_rsi_hw_interface
-- ~~  - moveit_calibration_gui
-- ~~  - wm_darknet_ros_3d_wrapper
-- ~~  - wm_frame_to_box
-- ~~  - yolo5_ros_msgs
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/kuka-ai/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kuka-ai/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:5990: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Thanks.

fgonzalezr1998, 2020-12-24:
Hi @ranjitkathiriya Have you cloned gb_visual_detection_3d (the melodic branch) and gb_visual_detection_3d_msgs (the melodic branch)?
I tell you it because the *master* branch contains de ROS2 version

ranjitkathiriya, 2021-01-05:
Thanks, I will check the branch and update you.

closed 2021-02-10 by ranjitkathiriya