IntelligentRoboticsLabs/gb_visual_detection_3d #44: Issues of working_frame

Litao917, 2021-01-20:
Hi developer,

Thanks for your amazing work. 
I was trying to crop the pointcloud2 published by realsense d435i in terms of the boundingbox_3d, so I changed the _working_frame_ in darknet_3d.yaml from camera_link to the camera_depth_optical_frame. But after that, I noticed that the boundingbox_markers in RVIZ are not precise as before. The boxes became larger than that with working_frame of camera_link. 

Any clues? Thank you in advance!

Litao917, 2021-01-20:
### Wrong boundboxes under working_frame of camera_depth_optical_frame
[image]
=======================================================================

### Correct boundboxes under working_frame of camera_link

[image]

Litao917, 2021-01-20:
[image]

Litao917, 2021-01-20:
@fgonzalezr1998

fmrico, 2021-01-22:
Hi!! 

Remember that this software requires that the X-axes points directly to the scene.

Litao917, 2021-01-23:
> Hi!!
> 
> Remember that this software requires that the X-axes points directly to the scene.

Hi fmrico,

After changed the working frame, the frame can be changed to the optical frame of Z-axis pointing out of the screen.

fgonzalezr1998, 2021-01-25:
Hi @Litao917  The algorithm of this package needs that the bounding boxes be calculated in a frame whose axes are like was described in previous issues. If you use a different frame, the bounding boxes will be calculated wrong.
So, if you need the bounding boxes in a different, frame, you must do a node that subscribes to the boxes and transform its coordinates to the new frame

Litao917, 2021-01-25:
> Hi @Litao917 The algorithm of this package needs that the bounding boxes be calculated in a frame whose axes are like was described in previous issues. If you use a different frame, the bounding [...]

Hi Fernando,
Thank you for your reply. Copy that. 
I will write a new node to transform them.

closed 2021-01-25 by Litao917