IntelligentRoboticsLabs/gb_visual_detection_3d #45: Send published coordinates to a python file

hsaleem1, 2021-01-25:
Hi there,

I am using this package and getting 3d coordinates in the following form:
  xmin: 0.4506256580352783
  ymin: -0.3164764642715454
  xmax: 0.7936256527900696
  ymax: 0.11368180811405182
  zmin: -0.25958430767059326
  zmax: 0.10506562888622284

I need to ask that can I save those coordinates so that I can call them in a python file?

Thanks for the help.

fgonzalezr1998, 2021-01-25:
Hi @hsaleem1 , what I would do is make a "fprintf" and execute the node with a shell redirection (">") to a file. Is it what you need?

hsaleem1, 2021-01-25:
Possibly this is a solution. Can you please guide me how to do that?

fgonzalezr1998, 2021-01-26:
@hsaleem1  You have a node in which you are subscribed to de bboxes 3d. Then, you get these coordinates and make a fprintf. More or less like that:
```
void callback(msg) {
  double xmin, ymin, zmin, xmax, ymax, zmax;
  xmin = ...;
  ymin = ...;
  zmin = ...;
  ...
  ...
  ...
  fprintf(stdout, "xmin: %lf\n", xmin);
  fprintf(stdout, "ymin: %lf\n", ymin);
  ...
  ...
}
```

hsaleem1, 2021-01-26:
@fgonzalezr1998, thanks for the help.

Could you please tell me that is there any way to save these coordinates into an excel file automatically?

hsaleem1, 2021-01-30:
@fgonzalezr1998, thank you very much.
Problem solved, and closing this issue.

closed 2021-01-30 by hsaleem1