IntelligentRoboticsLabs/gb_visual_detection_3d #46: Include wait time while getting coordinates.

hsaleem1, 2021-01-30:
The 3d coordinates of the detected objects are publishing continously. Is there any way to include a wait time through which there would be few seconds or milli seconds delay in publishing the 3d coordinates after the ealier detection?

Thanks.

fgonzalezr1998, 2021-02-12:
Yes! Unfortunately, you must change one thing of the code:

In the file **darknet3d_node.cpp** you will find the next line: ``ros::Rate loop_rate(10)``. That '10' is the frequency. You can change it by 5, 2, 20... what you want @hsaleem1

hsaleem1, 2021-02-17:
Hi @fgonzalezr1998 , I tried both increase and decrease the loop_rate but didn't see any difference. The coordinates are publishing continuously.

hsaleem1, 2021-02-18:
@fgonzalezr1998 , its done now. Actually, I did not do the catkin_make. Thanks a lot and closing the issue. Cheers!!!

closed 2021-02-18 by hsaleem1