IntelligentRoboticsLabs/gb_visual_detection_3d #48: Terminate called after throwing an instance of 'std::runtime_error'   what():  Time is out of dual 32-bit range

Prakadeeswaran05, 2021-05-21:
While trying to run this package in ros-melodic I face this error "terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range" and darknet 3d process dies but am able to see 2d detections. I am using a kinect camera attached to a mobile robot in gazebo. What could be the reason for this error?

I look forward to your response!
Prakadeeswaran M

ludovicoderic, 2021-06-02:
Same problem over here. Darknet_ros node launches, but the darknet_3d node dies with the same error code:
process[darknet_ros-1]: started with pid [14486]
process[darknet_3d-2]: started with pid [14487]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
[darknet_3d-2] process has died [pid 14487, exit code -6, cmd /home/ludo/catkin_ws/devel/lib/darknet_ros_3d/darknet3d_node __name:=darknet_3d __log:=/home/ludo/.ros/log/975599e8-c376-11eb-b4be-00155dad7402/darknet_3d-2.log].
log file: /home/ludo/.ros/log/975599e8-c376-11eb-b4be-00155dad7402/darknet_3d-2*.log

ludovicoderic, 2021-06-02:
Okay, I just solved the error, just have to follow this other issue request: https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/issues/36

And after changing the darknet_ros_3d/src/darknet_ros_3d/Darknet3D.cpp L64 from `last_detection_ts_ = ros::Time(now) - ros::Duration(60.0)` to `last_detection_ts_ = ros::Time(0)`, it now doenst gives that error, and it outputs the bounding boxes 3D correctly. 

But I will make some tests to see if that aproach is the correct way to do it.