IntelligentRoboticsLabs/gb_visual_detection_3d #49: cannot launch node of type [darknet_ros_3d/darknet3d_node]: Cannot locate node of type [darknet3d_node] in package [darknet_ros_3d].

ghokulji, 2021-05-21:
Hello,

I have cloned  the melodic branch of the darknet_ros_3d to my aws_robomaker  environment. I ran colcon build and the build was successful. When I run the darknet_ros_3d.launch I am facing this error.

[image]

> ERROR: cannot launch node of type [darknet_ros_3d/darknet3d_node]: Cannot locate node of type [darknet3d_node] in package [darknet_ros_3d]. Make sure file exists in package path and permission is set to executable (chmod +x)

I have tried the following approaches

I sourced my workspace with "source install/local_setup.sh" after colcon build. Apart from the darknet_ros_3d.launch, all other nodes are able to be launched. II also tried to run the node separately, but it seems like the executable cannot be found.
I got this error 

> [rosrun]` Couldn't find executable named darknet3d_node below /home/ubuntu/environment/aws_robomaker_hotbot/robot_ws/install/darknet_ros_3d/share/darknet_ros_3d

So, I thought the error must be with the CMakeLists.txt. But everything seems to be fine

> cmake_minimum_required(VERSION` 2.8.3)
> 
> project(darknet_ros_3d)
>
> set(CMAKE_BUILD_TYPE RelWithDebInfo)
>
> add_compile_options(-std=c++11)
> find_package(catkin REQUIRED COMPONENTS
>   roscpp
>   darknet_ros_msgs
>   gb_visual_detection_3d_msgs
>   sensor_msgs
>   tf2_ros
>   tf2_geometry_msgs
>   pcl_ros
>   pcl_conversions
>   roslint
> )
> 
> catkin_package(
>   CATKIN_DEPENDS ${PACKAGE_DEPENDENCIES}
>   DEPENDS PCL
>   INCLUDE_DIRS include
>   LIBRARIES ${PROJECT_NAME}
> )
> 
> include_directories(
>   include
>   ${catkin_INCLUDE_DIRS}
> )
> 
> add_library(${PROJECT_NAME}
>   src/darknet_ros_3d/Darknet3D.cpp
>   src/darknet_ros_3d/Darknet3DListener.cpp
>   src/darknet3d_node.cpp
> )
> 
> target_link_libraries(${PROJECT_NAME} ${catkin_LIBRARIES})
> 
> add_executable(darknet3d_node src/darknet3d_node.cpp)
> target_link_libraries(darknet3d_node ${catkin_LIBRARIES} ${PROJECT_NAME})
> 
> add_executable(darknet3d_listener_node test/darknet3d_listener_node.cpp)
> target_link_libraries(darknet3d_listener_node ${catkin_LIBRARIES} ${PROJECT_NAME})
> 
> 
> roslint_cpp(
>   src/darknet_ros_3d/Darknet3D.cpp include/darknet_ros_3d/Darknet3D.h
>   src/darknet_ros_3d/Darknet3DListener.cpp include/darknet_ros_3d/Darknet3DListener.h
>   src/darknet3d_node.cpp
> )
> 
> install(TARGETS ${PROJECT_NAME}
>    LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
> )
> 
> install(DIRECTORY include/${PROJECT_NAME}/
>      DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION})
> 
> install(DIRECTORY config launch 
>   DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
> )
> 

Can anyone help me out, as to why the executable is not found and why the build is not done properly?

Prakadeeswaran05, 2021-05-22:
Hey @ghokulji Have you tried this package in a gazebo environment ? If yes, could you help me with the steps? Currently am facing this error  "terminate called after throwing an instance of 'std::runtime_error' what(): Time is out of dual 32-bit range" .

fgonzalezr1998, 2021-05-27:
Hi @ghokulji  Did you compile the ROS package using colcon? You should do it using *catkin_make*. It may be the problem because in ROS1 you don't need to install the executables as ROS2 needs.

ghokulji, 2021-05-27:
Hello @fgonzalezr1998 , the aws robomaker environment uses colcon build to build the ROS1 packages also. During the build process, I am not facing any errors, it is only when executing the node.
I have tested the normal darkent_ros melodic and I built it using colcon build and it worked fine. To extract the depth of the detected object, I thought of using darknet_ros_3d. 

Can you recommend any changes that can be done to the CMakeLists.txt so that the build process creates an executable when using colcon build?

fgonzalezr1998, 2021-05-27:
Okey, so... The problem is that the executable *darknet3d_node* is not being installed. If you look the CMakeLists, you will see that the library and the launch file is being installed but not the executable. That is the error and this is the reason by it is not found. When I did this package I used catkin to compile and catkin doesn't need that.

If you solve this mistake, feel free to open a PR and I can put it in a new branch for other people that need it ;)

ghokulji, 2021-05-27:
@fgonzalezr1998, Thanks for the suggestion, I will try to add the install for the executables in CMakeLists.txt and see if it works out.

AbbiliKiran, 2023-01-28:
I need help resolving the following.ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure the file exists in the package path and permission is set to executable (chmod +x).

fgonzalezr1998, 2023-01-28:
Could you provide me more information? branch, steps you are following for launching the node, etc @AbbiliKiran