IntelligentRoboticsLabs/gb_visual_detection_3d #51: How can I get the xcenter, ycenter, z values..?

gusrbgks, 2021-07-13:
Hi. 

I think I've finished installing and running the darknet_3d package.

[image]
[image]

The darknet_ros_3d package outputs a 3D box.

As shown above, boundingBox3d message provides xmax, xmin, ymax, ymin, zmax, zmin .

But I only need the 3d coordinates of the center point of the 2D bbox detected by ros-darknet.

How can I get the xcenter, ycenter, z values..?

please answer about my question. Thank you.

gusrbgks, 2021-07-20:
Hi.
I seem to have found a way to get the 3D coordinates of the Center Point of a 2D bbox.
This is the center_point in this code.
is this right?

[image]

closed 2021-07-22 by gusrbgks

fgonzalezr1998, 2021-08-23:
Hi @gusrbgks First part is correct. However, the last part is not. You can not get the index in the point cloud from the (x,y) coordinates (At least, in this way) because that x and y are not pixels but they are 3D coordinates (in meters) taking the 'camera_link' (or what you use) frame.

gusrbgks, 2021-09-30:
Hi. @fgonzalezr1998 

The coordinates detected in Yolo were converted to the robot's base coordinates and moved, but they did not move to the desired coordinates.

I suspect it's because I put the wrong 3d coordinates.

So how can I get the 3D coordinates I want?