IntelligentRoboticsLabs/gb_visual_detection_3d #52: 3D point value is not output.

gusrbgks, 2021-07-22:
I got the center point (2D) of the bbox detected in Yolo. And I got a 3D point for this 2D point. So, I'm going to use this as a topic message.

I'm running it using roslaunch darknet_rod_3d darknet_ros_3d.launch.

I used this darknet_ros_3d while working on my project and there is a problem.

I made the 3D points output to the console window inside the code.

However, even if it is detected by Yolo, the 3D Point is not output.

So, I checked whether the topic is being output normally from rqt to the Topic Monitor, but it was not being output.

When I checked the checkbox of /darknet_ros_3d/bounding_boxes in Topic Monitor, 3D point was output normally.

What problem is this caused by..?

<before check>
[image]

<After check>
[image]

fgonzalezr1998, 2021-07-29:
I'm not sure if I have understood the problem. You have made your own project that consists of getting the center pixel and calculate its 3D coordinates, isn't it?
So... the problem is with your package or with darknet_3d? @gusrbgks

gusrbgks, 2021-08-06:
My guess is that there is something wrong with darknet_3d .

In order to use the obtained 3D coordinate values ​​in other nodes, the following command should be executed.

$ rostopic echo darknet_ros_3d/bounding_boxes

fgonzalezr1998, 2021-08-06:
Darknet ros  3d has a configuration file where the minimum probability is specified. If YOLO detections don't reach that probability, darknet 3d ignores them. This is probably what is happening. However, could you put here a fragment of your code where you get the data from the topic darknet_3d/bounding_boxes? @gusrbgks