IntelligentRoboticsLabs/gb_visual_detection_3d #53: std::out_of_range

cahidenes, 2021-07-28:
[image]
Machine: JetsonTX2
Linux version: Ubuntu 18.04
Ros distro: melodic
I wasn't able to increase point cloud point size which is 57600 (I think this is point cloud point size, please correct if it is not)

fgonzalezr1998, 2021-07-29:
What type of message are you subscribing to? You should subscribe to a *registered* point cloud. If you can put here the point cloud topic, it will be so helpfull @cahidenes

cahidenes, 2021-07-31:
I have an Intel D455 camera, and I was running rs_camera.launch. It looks like this option doesn't provide registered pointcloud. Running rs_rgbd.launch solved the issue. Thank you.

closed 2021-07-31 by cahidenes