IntelligentRoboticsLabs/gb_visual_detection_3d #70: Why sometimes bounding box marker doesn't appear in rviz?

TapleFlib, 2023-05-02:
The darknet_ros bounding box is there consistently, but the bounding_boxes_3d markers is not consistent and disappears frequently. Is minimum threshold the problem? I didn't change anything though.

fgonzalezr1998, 2023-05-15:
It may be due to different casues:
1. The probability decreases: In this case, try to decrease the minimun probability in the yaml
2. the minimum threshold is too small. Try changing it and see the results
3. Any kind of problem with the consistency of the point cloud: Monitor your point cloud for detecting any problem

These are the thing that are maybe causing some problems. If the problem persists after this trials, I would need more specific information about your system.

I hope my advices are useful @TapleFlib

TapleFlib, 2023-05-17:
I'll try it, thanks !

closed 2023-05-17 by TapleFlib