IntelligentRoboticsLabs/gb_visual_detection_3d #71: several errors

TapleFlib, 2023-06-03:
At first it can run normally, but after a few run, it shows the following error :

`at line 126 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
[ERROR] [1685775293.923868706]: Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty
Failed to find match for field 'x'.
Failed to find match for field 'y'.
Failed to find match for field 'z'.
Failed to find match for field 'rgb'.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 910) >= this->size() (which is 0)
handle: 100%
handle: 100%
handle: 100%
handle: 100%
handle: 100%
[darknet_3d-2] process has died [pid 70253, exit code -6, cmd /home/jay/project_ws/devel/lib/darknet_ros_3d/darknet3d_node __name:=darknet_3d __log:=/home/jay/.ros/log/79b8d08a-01d1-11ee-a442-2c56dcbb10f8/darknet_3d-2.log].
log file: /home/jay/.ros/log/79b8d08a-01d1-11ee-a442-2c56dcbb10f8/darknet_3d-2*.log
`

fgonzalezr1998, 2023-08-18:
It seems the tf used for the lookupTransform call is empty. What camera driver are you using? @TapleFlib

NET-Y, 2023-08-29:
Hello! I'm wondering if you have solved this problem yet, because I encountered the same problem.

TapleFlib, 2023-08-29:
> It seems the tf used for the lookupTransform call is empty. What camera driver are you using? @TapleFlib

I use realsense driver from realsense ros for realsense D455

TapleFlib, 2023-08-29:
> Hello! I'm wondering if you have solved this problem yet, because I encountered the same problem.

I have not found the main cause, but I can solve it by restarting the pc

NET-Y, 2023-08-30:
> > Hello! I'm wondering if you have solved this problem yet, because I encountered the same problem.
> 
> I have not found the main cause, but I can solve it by restarting the pc

Our group have confirmed that the problem originated from no publisher publishing pointclouds information ,but we have no idea how to solve the problem. 
[image]